Slamtec/rplidar_sdk #1: Change minimum distance

tiralonghipol, 2018-09-18:
Hi.
I am using the recent rplidar a3.
I would like to modify the minimum used range which I think is currently 0.15 m
How to do that? I tried to modify >range_min< inside the code (I'm using rplidar_ros) with no effect. 
Should I modify any parameter in the sdk ?

Thank you so much for the support!

cnwzhjs, 2018-10-30:
Hi, thank you for buying our product. Our SDK and ROS node do not have an embedded filtering feature. You may write a filter on your own.

tony-slamtec, 2019-05-05:
Each model of RPLIDAR has a physical limitation on the min distance (for A3, it's about 0.15 m), you cannot get measurement less than this physical limitation, but can filter out measurement if you want bigger min distance.

closed 2019-05-05 by tony-slamtec

BartVanderHaagen, 2024-06-03:
size_t m_scan_count = 0;
  double max_distance = 8.0f;
  double angle_min = deg_2_rad(0);
  double angle_max = deg_2_rad(359);
  const float min_distance = 0.15f;