Slamtec/rplidar_sdk #2: operation timed out on raspberry pi 3B

ameykasar, 2018-10-29:
I am trying to interface **RPLIDAR A1** with my **_Raspberry Pi 3 B 2015 Model running ubuntu 16.04_**.

I am receiving the following error when I run _ultra_simple /dev/ttyUSB0_: **Error, cannot bind to the specified serial port /dev/ttyUSB0.**

I receive the following error when I run _simple_grabber /dev/ttyUSB0_: **Error, operation timed out**

Error when I try interfacing using ROS (_rplidar_ros_ package): **RESULT_OPERATION_TIMEOUT** when I use _roslaunch rplidar_ros rplidar.launch_ 
I tried using multiple OS images but the error persists. It works fine on my PC running Ubuntu 16.04 with same USB cable and USB to serial converter.
What is causing these errors?
Is this due to Raspberry Pi 3 B only having USB 2.0 ports? Help me find a solution
--

cnwzhjs, 2018-10-30:
Have you ever checked the permission setting of the tty device? Maybe you can try to run the program under root account (or use sudo)

ameykasar, 2018-10-31:
@cnwzhjs
Now I am getting the following error:
```
process[rplidarNode-1]: started with pid [2362]
[ INFO] [1455209742.257557230]: RPLIDAR running on ROS package rplidar_ros. SDK Version:1.9.0
RPLIDAR S/N: BA4B9DF1C3E39AC4C3E698F94A9C340D
[ INFO] [1455209744.766681031]: Firmware Ver: 1.24
[ INFO] [1455209744.766873062]: Hardware Rev: 5
[ INFO] [1455209744.769167437]: RPLidar health status : 0
[ERROR] [1455209747.283383166]: Can not start scan: 80008000!

```
The LIDAR still works fine on my PC. Can this be a power issue? Is the raspberry pi unable to provide enough power? What can be done to ensure power is delivered?

ameykasar, 2019-02-11:
The problem is with cables. For future, ensure the cables are not dodgy and provide stable power for both the Pi and the Lidar.

closed 2019-02-11 by ameykasar

fidelDenzel, 2021-08-05:
If above solution didnt work, try checking the 7 pins cable between the usb converter and the uart lidar port. Try tightening them, unplug and plug it. Overall it is the matter of hardware solution, such as cable connectivity, power consumption, etc.