Slamtec/rplidar_sdk #6: Frame Grabber not working with RPLidar A1M1

Brooce, 2018-12-10:
We've got an RPLidar A1M1 device with 1.15 firmware - too old to upgrade, as stated by dfu_upgrader utility.

The ultra-simple app registers incoming data properly, i.e. the messages with theta and distance values are coming instantly. So the lidar works well.
However, nothing shows in the frame_grabber window, although the command buttons seem to work.

I noticed that there's an undocumented "welcome message" sent after power-up or restarting the device. It consists of 56 bytes, whereas some applications report it to be 63 in RPLidar A2 (which should have the same protocol as A1, according to the application's note). Could it be an issue here?

The "welcome message" that my lidar sends is (with CR+LF as newline characters):
>RP LIDAR System.
Firmware Ver 1.15, HW Ver 0
Mode: 0

esilky, 2019-03-19:
I recently bought a A1M8 and just received it. I have tried all three of the apps on three different machines:
1) Frame grabber
2) Simple grabber
3) Ultra-Simple
and (on all three machines) they report: 'cannot bind to the specified serial port com3'
The machine beeps when I plug/unplug, so I'm pretty sure it is recognizing the device, and the Lidar spins when I plug it in (and the green light on the bottom turns on).

Maybe the little serial adapter is bad?
Otherwise, I'll probably return the whole package.

johnelle, 2019-03-19:
Its been a while, but as I recall this software throws an error and looks
like dead but keeps running.  After tracking down a many years old compiler
to compile it, I finally got something that generated a point cloud (with
errors).

Like most Chinese hardware the software is buggy and pretty much unusable
with no documentation or support.

Unless you want to code to bare bones through experimentation you may want
to return it.

204maker, 2019-06-05:
I just recently bought an A1M8. The demo solution file throws a ton of errors for VS 2019 Community edition. Has anyone used these successfully recently?

360sepehr, 2020-03-27:
> I recently bought a A1M8 and just received it. I have tried all three of the apps on three different machines:
>
> 1. Frame grabber
> 2. Simple grabber
> 3. Ultra-Simple
>    and (on all three machines) they [...]

Hi sir
Could you solve the problem of launching Lidar?
My Lidar doesn't even know the computer port

fidelDenzel, 2021-07-23:
If the computer doesnt detect the serial, then just download the manual guide at the website given.
But if the usb was already assigned to COM_ , then just run the frame grabber and mine work with 57.6 k baud rate with Windows

NouriHuseynov, 2021-09-06:
Hi everyone,
if you read this post, it will significate that you want to communicate with LiDAR in order to integrate it to your project. So, I thinks that my work can interest you, check it at this link : Use the LiDAR A1M8 
BR, 
chucknouri

Marsfan, 2021-09-06:
> 
> 
> Hi everyone,
> if you read this post, it will significate that you want to communicate with LiDAR in order to integrate it to your project. So, I thinks that my work can interest you, check it at this link : [https://github.com/NouriHuseynov/LiDAR_A1M8.git ](https://github.com/NouriHuseynov/LiDAR_A1M8.git%C2%A0)
> BR,
> Nouri Huseynov

Hey. I just waned to let you know that the link to your project is broken, you need to remove the `%C2%A0` from the end of the link. 

Sadly, I no longer have access my a RPLIDAR, or I would love to help out.

NouriHuseynov, 2021-09-06:
Oh sorry @Marsfan I update the link in my post, I think everything is ok now (I hope)...

Marsfan, 2021-09-06:
@NouriHuseynov Yup. Looks good. No worries, just wanted to make sure you know about it