Slamtec/rplidar_sdk #12: 100% CPU

xrefactory, 2019-03-13:
Hi, 

I am working on Raspberry Pi and Linux. I have written a small program using rplidar sdk. Usually it works perfectly and everything is ok. However, if I unplug lidar's usb connector while the program is running then the program starts to consume 100%. CPU is not consumed in the main thread, it is consumed in a thread created by rplidar's sdk. Any help?

Best regards,

Marian

tony-slamtec, 2019-05-05:
It is caused by serial port code in the SDK, when the serial port doesn't exist anymore, the calls will return error immediately which causes the loop consuming all CPU.

We can improve this in latter release. For a workaround, please destruct the driver when you receives error from driver, and try reconnect.

NouriHuseynov, 2021-09-06:
Hi everyone,
if you read this post, it will significate that you want to communicate with LiDAR in order to integrate it to your project. So, I thinks that my work can interest you, check it at this link : https://github.com/NouriHuseynov/LiDAR_A1M8.git
BR,
Nouri Huseynov