Slamtec/rplidar_sdk #13: Better documentation for `scan_mode` launch parameter

Alescontrela, 2019-04-02:
For some applications, it's important to have control of the sample rate of the LiDAR. The RPLidar a3 supports different scan modes, yet these different scan modes are not well documented.

Please update the wiki to include the different scan modes. So far, this is what I have found:

scan_mode dictates the RPLIDAR A3's sample rate
* "Boost" => 8000 samples
* "Sensitivity" => 4000 samples
* "Express" => 2000 samples

rotu, 2019-07-15:
Agreed. It's not clear from the SDK that they affect anything other than the max distance and sample rate. I did find some explanation in the [Interface Protocol and Application Notes](http://bucket.download.slamtec.com/ccb3c2fc1e66bb00bd4370e208b670217c8b55fa/LR001_SLAMTEC_rplidar_protocol_v2.1_en.pdf) figure 3-3 page 12. Notably:

* Boost: Optimized for sample rate
* Sensitivity: Optimized for longer range, better sensitivity, but weak environment light elimination.
* Stability: Optimized for environment light elimination performance, but shorter range and lower sample rate.

HourGlss, 2023-09-10:
what are the "scan mode ids" so that I can correctly pack an extended version EXPRESS_SCAN?