Slamtec/rplidar_sdk #14: Cannot compile using clang

drdanz, 2019-04-11:
Build using clang fails with a couple of `-Wc++11-narrowing` errors:

```
make[1]: Entering directory '/tmp/rplidar_sdk/sdk/sdk'
 CXX  src/rplidar_driver.cpp
src/rplidar_driver.cpp:1786:10: error: case value evaluates to -1, which cannot be narrowed to type 'unsigned long' [-Wc++11-narrowing]
    case rp::hal::Event::EVENT_TIMEOUT:
         ^
src/rplidar_driver.cpp:1815:10: error: case value evaluates to -1, which cannot be narrowed to type 'unsigned long' [-Wc++11-narrowing]
    case rp::hal::Event::EVENT_TIMEOUT:
         ^
2 errors generated.
make[1]: *** [/tmp/rplidar_sdk/sdk/mak_common.inc:82: /tmp/rplidar_sdk/sdk/obj/Linux/Release/sdk/src/rplidar_driver.o] Error 1
make[1]: Leaving directory '/tmp/rplidar_sdk/sdk/sdk'
make: *** [/tmp/rplidar_sdk/sdk/mak_common.inc:43: make_subs] Error 1
```

Alko89, 2019-06-06:
I'm having the same issue trying to cross compile for Android.

JensC, 2020-04-06:
me too on macOS

bartekpacia, 2020-12-21:
same issue, I'd be grateful if someone shared how they solved it

l376571926, 2022-07-25:
> I'm having the same issue trying to cross compile for Android.

case (unsigned long)rp::hal::Event::EVENT_TIMEOUT:

l376571926, 2022-07-25:
> case (unsigned long)rp::hal::Event::EVENT_TIMEOUT:

case (unsigned long)rp::hal::Event::EVENT_TIMEOUT: