Slamtec/rplidar_sdk #18: Demo solution does not compile for Visual Studio 2019 Community edition

204maker, 2019-06-05:
Are the codes for the SDK being still being maintained?

They do not seem to compile on Visual Studio 2019 Community edition and it shows a lot of warnings/errors. It is year 2019 at the moment, do you expect us to compile the source codes for the demo solution using 2009/2010 version of Visual Studio?

dvillevald, 2019-07-31:
It does not compile on Visual Studio 2017 Community

johnelle, 2019-07-31:
It doesn't compile PERIOD.  Even on the specified version of VS (which is a decade old) you get lots of errors.  It does produce a couple of images that run with errors, but its basically a Chinese dump-and-run product.  The assumption is that any working code is written by the purchaser with no documentation, working examples, etc.

dvillevald, 2019-08-03:
I contacted the Amazon vendor (seeed.cc) I purchased RPLIDAR A1 from - below is the vendor's response: 

_We checked with Slamtech design team and they said they currently support  Microsoft Visual C++ 2010. They do not have plans to upgrade to 2017 at this moment._

justin-sprunger, 2019-10-08:
I was able to download VS2010 ultimate from CNET of all places. From there I was able to compile and use the Frame_Grabber application. I have not tried any newer versions of VS, but VS2010 does work and luckily a fully functional demo still lives!