Slamtec/rplidar_sdk #21: Compilation issues on MacOS for simple_grabber (Mojave)

apsamuel, 2019-06-27:
Running into an issue compiling the SDK on my mac (details on build environment below). Please let me know if you can provide any assistance. 

**Build Environment** 

```
~/DevOps/Applications/rplidar_sdk/sdk(local*) » uname -a
Darwin triplehelix 18.6.0 Darwin Kernel Version 18.6.0: Thu Apr 25 23:16:27 PDT 2019; root:xnu-4903.261.4~2/RELEASE_X86_64 x86_64
```

```
~/DevOps/Applications/rplidar_sdk/sdk(local*) » sw_vers                                                                                                     photon@triplehelix-2
ProductName:	Mac OS X
ProductVersion:	10.14.5
BuildVersion:	18F132
```

```
~/DevOps/Applications/rpilidarsdk/sdk(v1.11.0*) » clang --version
Apple LLVM version 10.0.1 (clang-1001.0.46.4)
Target: x86_64-apple-darwin18.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

**Build Error** 

```
~/DevOps/Applications/rpilidarsdk(master) » git checkout -b v1.11.0 release/v1.11.0
Switched to a new branch 'v1.11.0'

~/DevOps/Applications/rpilidarsdk/sdk(v1.11.0) » make
 CXX  src/rplidar_driver.cpp
 CXX  src/hal/thread.cpp
 CXX  src/arch/macOS/net_serial.cpp
 CXX  src/arch/macOS/net_socket.cpp
 CXX  src/arch/macOS/timer.cpp
mkdir -p `dirname /Users/photon/DevOps/Applications/rpilidarsdk/sdk/output/Darwin/Release/librplidar_sdk.a`
 pack rplidar_driver.o->librplidar_sdk.a
 pack thread.o->librplidar_sdk.a
 pack net_serial.o->librplidar_sdk.a
 pack net_socket.o->librplidar_sdk.a
 pack timer.o->librplidar_sdk.a
 CXX  main.cpp
main.cpp:118:16: warning: 'grabScanData' is deprecated [-Wdeprecated-declarations]
    ans = drv->grabScanData(nodes, count);
               ^
/Users/photon/DevOps/Applications/rpilidarsdk/sdk/app/simple_grabber/../../sdk/include/rplidar_driver.h:257:5: note: 'grabScanData' has been explicitly marked deprecated here
    DEPRECATED(virtual u_result grabScanData(rplidar_response_measurement_node_t * nodebuffer, size_t & count, _u32 timeout = DEFAULT_TIMEOUT)) = 0;
    ^
/Users/photon/DevOps/Applications/rpilidarsdk/sdk/app/simple_grabber/../../sdk/include/rplidar_driver.h:44:55: note: expanded from macro 'DEPRECATED'
        #define DEPRECATED(func) func __attribute__ ((deprecated))
                                                      ^
main.cpp:120:14: warning: 'ascendScanData' is deprecated [-Wdeprecated-declarations]
        drv->ascendScanData(nodes, count);
             ^
/Users/photon/DevOps/Applications/rpilidarsdk/sdk/app/simple_grabber/../../sdk/include/rplidar_driver.h:285:5: note: 'ascendScanData' has been explicitly marked deprecated here
    DEPRECATED(virtual u_result ascendScanData(rplidar_response_measurement_node_t * nodebuffer, size_t count)) = 0;
    ^
/Users/photon/DevOps/Applications/rpilidarsdk/sdk/app/simple_grabber/../../sdk/include/rplidar_driver.h:44:55: note: expanded from macro 'DEPRECATED'
        #define DEPRECATED(func) func __attribute__ ((deprecated))
                                                      ^
main.cpp:42:20: warning: unused function 'delay' [-Wunused-function]
static inline void delay(_word_size_t ms){
                   ^
3 warnings generated.
 LD   /Users/photon/DevOps/Applications/rpilidarsdk/sdk/output/Darwin/Release/simple_grabber
ld: warning: ignoring file /Users/photon/DevOps/Applications/rpilidarsdk/sdk/output/Darwin/Release/librplidar_sdk.a, file was built for archive which is not the architecture being linked (x86_64): /Users/photon/DevOps/Applications/rpilidarsdk/sdk/output/Darwin/Release/librplidar_sdk.a
Undefined symbols for architecture x86_64:
  "rp::standalone::rplidar::RPlidarDriver::CreateDriver(unsigned int)", referenced from:
      _main in main.o
  "rp::standalone::rplidar::RPlidarDriver::DisposeDriver(rp::standalone::rplidar::RPlidarDriver*)", referenced from:
      _main in main.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/photon/DevOps/Applications/rpilidarsdk/sdk/output/Darwin/Release/simple_grabber] Error 1
make[1]: *** [make_subs] Error 1
make: *** [make_subs] Error 1
```

```
~/DevOps/Applications/rpilidarsdk/sdk(v1.11.0*) » lipo -detailed_info output/Darwin/Release/librplidar_sdk.a
input file output/Darwin/Release/librplidar_sdk.a is not a fat file
Non-fat file: output/Darwin/Release/librplidar_sdk.a is architecture: x86_64
```

craiganderson-53546, 2019-11-02:
I have the same issue on Linux (Cygwin actually).  Not only that but after successfully compiling using VS 2010, running the resulting simple_grabber.exe displays the same deprecated warning then hangs.  Have you found any solution?

craiganderson-53546, 2019-11-09:
Not sure why the code is delivered in a state where it uses functions that are deprecated, has error messages to the effect, and also contains the new functions.  Anyway, the fix is to upgrade the calls to the new functions, new variables required by the functions, and updated print statements to output the correct data...

```
    rplidar_response_measurement_node_hq_t nodes[8192];  // .._hq_t
    size_t   count = _countof(nodes);

    printf("waiting for data...\n");
    // fetch exactly one 0-360 degrees' scan
    ans = drv->grabScanDataHq(nodes, count);  //  ..DataHq
    if (IS_OK(ans) || ans == RESULT_OPERATION_TIMEOUT) {
        drv->ascendScanData(nodes, count);
        plot_histogram(nodes, count);

        printf("Do you want to see all the data? (y/n) ");
        int key = getchar();
        if (key == 'Y' || key == 'y') {
            for (int pos = 0; pos < (int)count ; ++pos) {
                printf("%s theta: %03.2f Dist: %08.2f \n", 
                    (nodes[pos].quality & RPLIDAR_RESP_MEASUREMENT_SYNCBIT) ?"S ":"  ", 
                    (nodes[pos].angle_z_q14  >> RPLIDAR_RESP_MEASUREMENT_ANGLE_SHIFT)/64.0f,  // angle_z_q14
                    nodes[pos].dist_mm_q2/4.0f);  //  dist_mm_q2
            }
        }
```