Slamtec/rplidar_sdk #24: 请问一下提出数据后怎么转成float等基本类型

gentle-li, 2019-08-10:
我们在取出角度数据后，使用cos函数会报错，但是cos函数使用其他数据就不会报错，怀疑是数据类型的问题。请问一下怎么解决。

vinjn, 2019-08-11:
建议在出错的地方设断点，查看是否是 `inf` / `nan` 等异常数据

gentle-li, 2019-08-13:
我是在树莓派上运行的，在编译的时候就出错了，我尝试过cos（PI）是可以编译的， 但是cos（theta）不行，PI是float的3.1415926  theta是原始代码打印在终端的theta

gentle-li, 2019-08-13:
mkdir -p `dirname /home/pi/Downloads/rplidar_sdk-master/sdk/output/Linux/Release/librplidar_sdk.a`
 pack main.o->librplidar_sdk.a
 LD   /home/pi/Downloads/rplidar_sdk-master/sdk/output/Linux/Release/ultra_simple
/usr/bin/ld: /home/pi/Downloads/rplidar_sdk-master/sdk/obj/Linux/Release/ultra_simple/main.o: undefined reference to symbol 'cosf@@GLIBC_2.4'
//lib/arm-linux-gnueabihf/libm.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
/home/pi/Downloads/rplidar_sdk-master/sdk/mak_common.inc:72: recipe for target '/home/pi/Downloads/rplidar_sdk-master/sdk/output/Linux/Release/ultra_simple' failed
make[2]: *** [/home/pi/Downloads/rplidar_sdk-master/sdk/output/Linux/Release/ultra_simple] Error 1
make[2]: Leaving directory '/home/pi/Downloads/rplidar_sdk-master/sdk/app/ultra_simple'
/home/pi/Downloads/rplidar_sdk-master/sdk/mak_common.inc:43: recipe for target 'make_subs' failed
make[1]: *** [make_subs] Error 1
make[1]: Leaving directory '/home/pi/Downloads/rplidar_sdk-master/sdk/app'
/home/pi/Downloads/rplidar_sdk-master/sdk/mak_common.inc:43: recipe for target 'make_subs' failed
make: *** [make_subs] Error 1

gentle-li, 2019-08-13:
谢谢，已经弄好啦，请问一下distance的单位是什么

l376571926, 2022-07-26:
> 谢谢，已经弄好啦，请问一下distance的单位是什么

https://bucket-download.slamtec.com/0e62eb7b7b3bd207d6973591a71b68d2a5f47fde/LR001_SLAMTEC_rplidar_protocol_v2.4_cn.pdf