Slamtec/rplidar_sdk #28: Added support for MCU

wuhanstudio, 2019-10-04:
I noticed that there is only one bare minimum example for STM32. It sends out scan command and then prints out response as raw data without parsing.

https://wiki.slamtec.com/pages/viewpage.action?pageId=13959187

In order to use RPLidar on MCUs, I ported the SDK to RT-Thread which works on STM32, RISC-V, i.mx RT, C-SKY, virtually all mainstream MCUs.

https://github.com/wuhanstudio/rplidar

I hope this will make RTLidar working MCUs more feasible.

-----------------------
Han Wu, Master Student
School of Energy and Power Engineering
Huazhong University of Science and Technology
Email: wuhanstudio@hust.edu.cn
Website: http://wuhanstudio.cc

Vishnu-Vijay95, 2021-07-21:
This works for both RP LIDAR A1 and A2? while interfacing with the MCU, the PWM input to LIDAR motor is to be carried out separately?

wuhanstudio, 2021-07-21:
Hi, this should work for both A1 and A2 as they share the same protocol:

http://bucket.download.slamtec.com/f010c72be308cdc618e91746d643278185ed02b2/LR001_SLAMTEC_rplidar_protocol_v2.2_en.pdf

But this package is only tested on A1, because I do not currently have a A2 for testing.

The motor can be controlled using the serial protocol. thus only UART is required for the connection.

Vishnu-Vijay95, 2021-07-21:
I had gone through the code, i believe you have written it for RT-thread RTOS ? I am aiming to use the lidar driver program for a project in freeRTOS on a stm32 hardware. I think i need to port  the rplidar.c file to freeRTOS, right?

wuhanstudio, 2021-07-21:
The serial driver that requires porting is here:

https://github.com/wuhanstudio/rplidar/blob/master/drv/drv_rplidar.c

And the serial protocol is implemented here:

https://github.com/wuhanstudio/rplidar/blob/master/src/rplidar.c