Slamtec/rplidar_sdk #31: I'm concerned that there are no updates

tkircher, 2020-03-07:
I want to keep using Slamtec products, but it doesn't appear there's much in the way of support. They do function reasonably well, but both Microsoft and Intel have LIDAR products now. This repository has not been updated in many months and there don't appear to be any efforts to respond to any bug reports or handle pull requests.

bramburn, 2020-03-07:
Hi you need to email them it’s faster
I don’t understand why you need to see the code being updated constantly. It’s working fine for me.
I had an email response within hours

tkircher, 2020-03-07:
It's working fine for me too. I don't have any specific issue that I personally need to have solved. The purpose of public updates is for transparency and community engagement, two things that build trust in companies.