Slamtec/rplidar_sdk #38: ultra_simple 1.12.0 doesnt run on Linux with RPLidar S1

MegaraSoftware, 2020-04-19:
The sample tool ultra_simple version 1.12.0 doesn't run on Linux with RPLidar S1
version 1.11.0 work well.

Version 1.12.0
[image]

Version 1.11.0
[image]

tkircher, 2020-04-25:
Yeah, this isn't being maintained at all.

GerwinV16, 2020-08-17:
Had the same issue, seems the fix is pretty simple: add " return RESULT_OK;" to the startMotor() function after line 2222 in the file rplidar_drver.cpp so it returns a proper u_result.
This seems fixed in the ros version of the library but not in this one.

MegaraSoftware, 2020-12-16:
Thanks, yes this is the right fix.

esamet, 2022-10-24:
hi ı have a this error ı use ubuntu 20.04 and rplidar s1 with usb adapter Error, cannot bind to the specified serial port /dev/ttyUSB0.

ezhovv, 2024-01-23:
@esamet how did you resolve the issue? facing the thing now

deyouslamtec, 2024-01-24:
@ezhovv It may be/dev/ttyUSB0 without reading and writing permissions