Slamtec/rplidar_sdk #45: Error, cannot bind to the specified serial port /dev/ttyUSB0

VbsmRobotic, 2020-11-07:
Hi, 
I am using ubuntu 18.04.  I have already root privilege to access tty device as below :
    $ cd /etc/udev/rules.d/   \\
    $ sudo touch 50-usb-serial.rules   \\
    $ sudo gedit 50-usb-serial.rules   \\
In the file that is opened by the GEdit, wrote these words: KERNEL=="ttyUSB0", GROUP="vahid", MODE="0666" 
Also run 
$ sudo chmod 0666 /dev/ttyUSB0

When I run ultra_simple, face issue as follow : 
~/rplidar_sdk/sdk/output/Linux/Release$ sudo ./ultra_simple /dev/ttyUSB0 115200

Ultra simple LIDAR data grabber for RPLIDAR.
Version: 1.12.0
Error, cannot bind to the specified serial port /dev/ttyUSB0.

Anyone has face this issue. Please advise

pirate999, 2020-12-09:
I have the same issue, have you solved ?

VbsmRobotic, 2020-12-09:
Please change buad rate to 256000. It should work.

PauloSilva98, 2020-12-23:
Did you manage to solve this issue? Same thing is happening to me...

VbsmRobotic, 2020-12-23:
> Did you manage to solve this issue? Same thing is happening to me...

Have changed buad rate to 256000 ? 
follow  https://github.com/Slamtec/rplidar_ros

PauloSilva98, 2020-12-23:
Yes I have, both in the RP Lidar itself and the when running it

VbsmRobotic, 2020-12-23:
> Yes I have, both in the RP Lidar itself and the when running it

How about ttyUSB permission ?

PauloSilva98, 2020-12-23:
> > Yes I have, both in the RP Lidar itself and the when running it
> 
> How about ttyUSB permission ?

KERNEL=="ttyUSB0", GROUP="vahid", MODE="0666"

I have added this into a new rules file in the udev

VbsmRobotic, 2020-12-23:
> > > Yes I have, both in the RP Lidar itself and the when running it
> > 
> > 
> > How about ttyUSB permission ?
> 
> KERNEL=="ttyUSB0", GROUP="vahid", MODE="0666"
> 
> I have added this into a new rules file in the udev

Please remove GROUP= "vahid" after that from terminal run   $ sudo chmod 666 /dev/ttyUSB0 based on your usb port  select ttyUSB0 or ttyUSB1 ...

PauloSilva98, 2020-12-23:
> Please remove GROUP= "vahid" after that from terminal run $ sudo chmod 666 /dev/ttyUSB0 based on your usb port select ttyUSB0 or ttyUSB1 ...

Tried it still doesn't work. This is what pops up with this command:
$ ls -l /dev |grep ttyUSB
crw-rw-rw-  1 root        root    188,     0 dez 23 10:57 ttyUSB0

So i see that it is in ttyUSB0 (and it has never changed). I don't understand why i still don't have permission. Is this "crw-rw-rw" not enough?

J0ANMM, 2021-07-30:
I had a similar issue and the following command worked for me: `sudo adduser $USER dialout`
I found this solution [here](https://answers.ros.org/question/199561/rplidar-driver-installation-problem-on-ubuntu/?answer=199598#post-id-199598).

bensonhi, 2022-10-14:
Ha

kadenslater95, 2023-09-30:
Hello everyone, I ran into this problem and it was a pain trying everything they mentioned above. Finally I figured it out. First, you do need to get the permissions set up right so do follow at least one of those suggestions above. Now with that out of the way, what was causing the issue is that the default Baudrate, and the suggested above Baudrate of 265000 were incorrect for my particular device. I have the model A1M8. It turns out that if you just type `./ultra_simple` on the latest version of the sdk from github it will print out the usage which includes a list of baud rates. I had to use the A1 baudrate of 115200 and voila, it works now.

For reference here is the list that spits out in the terminal
A1(115200),A2M7(256000),A2M8(115200),A2M12(256000),A3(256000),S1(256000),S2(1000000),S3(1000000)

kadenslater95, 2023-09-30:
One final note, you should update the README with this little tricky configuration issue and mark this issue as closed