Slamtec/rplidar_sdk #50: won't compile on debian linux

bsautner, 2021-03-15:
I just bought a fancy Slamtec RPLIDAR A1 - 360 Laser Range Scanner from adafruit 

this led me here to clone and make this project trying to run the ultra simple demo - 

in the sdk dir i run make - i have g++ installed - errors below error code was:

make: *** [/home/ben/code/rplidar_sdk/sdk/mak_common.inc:43: make_subs] Error 1

```
ake[1]: Entering directory '/home/ben/code/rplidar_sdk/sdk/sdk'
 CXX  src/rplidar_driver.cpp
src/rplidar_driver.cpp: In member function ‘virtual int rp::standalone::rplidar::RPlidarDriverImplCommon::_getSyncBitByAngle(int, int)’:
src/rplidar_driver.cpp:671:16: warning: variable ‘last_angleInc_q16’ set but not used [-Wunused-but-set-variable]
  671 |     static int last_angleInc_q16 = 0;
      |                ^~~~~~~~~~~~~~~~~
src/rplidar_driver.cpp:673:9: warning: unused variable ‘syncBit_check_threshold’ [-Wunused-variable]
  673 |     int syncBit_check_threshold = (int)((5 << 16) / angleInc_q16) + 1;//find syncBit in 0~3 degree
      |         ^~~~~~~~~~~~~~~~~~~~~~~
src/rplidar_driver.cpp: In member function ‘virtual u_result rp::standalone::rplidar::RPlidarDriverImplCommon::_cacheUltraCapsuledScanData()’:
src/rplidar_driver.cpp:771:46: warning: unused variable ‘last_scan_count’ [-Wunused-variable]
  771 |     size_t                                   last_scan_count = 0;
      |                                              ^~~~~~~~~~~~~~~
src/rplidar_driver.cpp: In member function ‘virtual void rp::standalone::rplidar::RPlidarDriverImplCommon::_capsuleToNormal(const rplidar_response_capsule_measurement_nodes_t&, rplidar_response_measurement_node_hq_t*, size_t&)’:
src/rplidar_driver.cpp:850:17: warning: unused variable ‘syncBit_check_threshold’ [-Wunused-variable]
  850 |             int syncBit_check_threshold = (int)((2 << 16) / angleInc_q16) + 1;//find syncBit in 0~1 degree
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
src/rplidar_driver.cpp: In function ‘_u32 rp::standalone::rplidar::_crc32cal(_u32, void*, _u16)’:
src/rplidar_driver.cpp:1027:23: warning: suggest parentheses around ‘-’ in operand of ‘&’ [-Wparentheses]
 1027 |     _u8 leftBytes = 4 - len & 0x3;
      |                     ~~^~~~~
src/rplidar_driver.cpp: In member function ‘virtual void rp::standalone::rplidar::RPlidarDriverImplCommon::_ultraCapsuleToNormal(const rplidar_response_ultra_capsule_measurement_nodes_t&, rplidar_response_measurement_node_hq_t*, size_t&)’:
src/rplidar_driver.cpp:1238:32: warning: comparison of integer expressions of different signedness: ‘int’ and ‘unsigned int’ [-Wsign-compare]
 1238 |             if ((dist_predict1 == 0xFFFFFE00) || (dist_predict1 == 0x1FF)) {
      |                  ~~~~~~~~~~~~~~^~~~~~~~~~~~~
src/rplidar_driver.cpp:1246:32: warning: comparison of integer expressions of different signedness: ‘int’ and ‘unsigned int’ [-Wsign-compare]
 1246 |             if ((dist_predict2 == 0xFFFFFE00) || (dist_predict2 == 0x1FF)) {
      |                  ~~~~~~~~~~~~~~^~~~~~~~~~~~~
src/rplidar_driver.cpp:1256:21: warning: unused variable ‘syncBit_check_threshold’ [-Wunused-variable]
 1256 |                 int syncBit_check_threshold = (int)((3 << 16) / angleInc_q16)+1;//find syncBit in 0~1 degree
      |                     ^~~~~~~~~~~~~~~~~~~~~~~
src/rplidar_driver.cpp: In member function ‘virtual u_result rp::standalone::rplidar::RPlidarDriverImplCommon::grabScanData(rplidar_response_measurement_node_t*, size_t&, _u32)’:
src/rplidar_driver.cpp:1826:26: error: narrowing conversion of ‘rp::hal::Event::EVENT_TIMEOUT’ from ‘int’ to ‘long unsigned int’ [-Wnarrowing]
 1826 |     case rp::hal::Event::EVENT_TIMEOUT:
      |                          ^~~~~~~~~~~~~
src/rplidar_driver.cpp: In member function ‘virtual u_result rp::standalone::rplidar::RPlidarDriverImplCommon::grabScanDataHq(rplidar_response_measurement_node_hq_t*, size_t&, _u32)’:
src/rplidar_driver.cpp:1855:26: error: narrowing conversion of ‘rp::hal::Event::EVENT_TIMEOUT’ from ‘int’ to ‘long unsigned int’ [-Wnarrowing]
 1855 |     case rp::hal::Event::EVENT_TIMEOUT:
      |                          ^~~~~~~~~~~~~
src/rplidar_driver.cpp: In member function ‘virtual u_result rp::standalone::rplidar::RPlidarDriverImplCommon::startMotor()’:
src/rplidar_driver.cpp:2222:26: warning: control reaches end of non-void function [-Wreturn-type]
 2222 |         setLidarSpinSpeed(600);//set default rpm to tof lidar
      |         ~~~~~~~~~~~~~~~~~^~~~~
make[1]: *** [/home/ben/code/rplidar_sdk/sdk/mak_common.inc:82: /home/ben/code/rplidar_sdk/sdk/obj/Linux/Release/sdk/src/rplidar_driver.o] Error 1
make[1]: Leaving directory '/home/ben/code/rplidar_sdk/sdk/sdk'
make: *** [/home/ben/code/rplidar_sdk/sdk/mak_common.inc:43: make_subs] Error 1

```

focusrobotics, 2021-03-16:
I just hit the same error. Making this change to event.h got it to compile and the demos are now running properly for me:

diff --git a/sdk/sdk/src/hal/event.h b/sdk/sdk/src/hal/event.h
index 5e05234..f25d5ac 100644
--- a/sdk/sdk/src/hal/event.h
+++ b/sdk/sdk/src/hal/event.h
@@ -42,7 +42,7 @@ public:
     enum
     {
         EVENT_OK = 1,
-        EVENT_TIMEOUT = -1,
+        EVENT_TIMEOUT = 2,
         EVENT_FAILED = 0,
     };

bsautner, 2021-03-16:
Thanks - i ended up tossing out the rplidar USB adapter which also seemed defective on arrival and used an arduino to read from the RPLIDAR which worked out - there are several libraries out there and you just need a serial port

xayhewalo, 2022-03-14:
For people coming to this error now, the current code in master has `0xFFFFFFFFFFFFFFFF` for the value of `EVENT_TIMEOUT` instead of `-1`. However, _**v1.12.0 and below still contain this error**_ (or at least v1.11.0, the version on slamtech.com, still does).

In short, if you're compiling on a Raspberry Pi: clone master, don't use the release tags.

https://github.com/Slamtec/rplidar_sdk/blob/9d85aae434d39fc4014018dcbf8711f8918b0254/sdk/src/hal/event.h#L46

xayhewalo, 2022-03-14:
For those curious the offending line in master currently has `EVENT_TIMEOUT` set to `0xFFFFFFFFFFFF` instead of `-1` but that's still giving me problems on my RPI 3A+. No problems on Ubuntu 20.04 though... Setting the enum to `2` fixes the issue. Haven't figured out why. #54 probably fixes this issue.