Slamtec/rplidar_sdk #55: Make does not sucessfully make SDK

GasserJC, 2021-04-28:
Here is what happens when I run Make,
After this, I add the include directory in my source code, and the sdk runs into a hal/types.h error - stating this is not being linked during the linkage of the files. I also attempting referencing all sdk src files within the g++ compilation and this does not help.

```
make[1]: Entering directory '/home/jacob/SourceCode/LiDAR/rplidar_sdk/sdk/sdk'
 CXX  src/rplidar_driver.cpp
src/rplidar_driver.cpp: In member function ‘virtual int rp::standalone::rplidar::RPlidarDriverImplCommon::_getSyncBitByAngle(int, int)’:
src/rplidar_driver.cpp:671:16: warning: variable ‘last_angleInc_q16’ set but not used [-Wunused-but-set-variable]
  671 |     static int last_angleInc_q16 = 0;
      |                ^~~~~~~~~~~~~~~~~
src/rplidar_driver.cpp:673:9: warning: unused variable ‘syncBit_check_threshold’ [-Wunused-variable]
  673 |     int syncBit_check_threshold = (int)((5 << 16) / angleInc_q16) + 1;//find syncBit in 0~3 degree
      |         ^~~~~~~~~~~~~~~~~~~~~~~
src/rplidar_driver.cpp: In member function ‘virtual u_result rp::standalone::rplidar::RPlidarDriverImplCommon::_cacheUltraCapsuledScanData()’:
src/rplidar_driver.cpp:771:46: warning: unused variable ‘last_scan_count’ [-Wunused-variable]
  771 |     size_t                                   last_scan_count = 0;
      |                                              ^~~~~~~~~~~~~~~
src/rplidar_driver.cpp: In member function ‘virtual void rp::standalone::rplidar::RPlidarDriverImplCommon::_capsuleToNormal(const rplidar_response_capsule_measurement_nodes_t&, rplidar_response_measurement_node_hq_t*, size_t&)’:
src/rplidar_driver.cpp:850:17: warning: unused variable ‘syncBit_check_threshold’ [-Wunused-variable]
  850 |             int syncBit_check_threshold = (int)((2 << 16) / angleInc_q16) + 1;//find syncBit in 0~1 degree
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
src/rplidar_driver.cpp: In function ‘_u32 rp::standalone::rplidar::_crc32cal(_u32, void*, _u16)’:
src/rplidar_driver.cpp:1027:23: warning: suggest parentheses around ‘-’ in operand of ‘&’ [-Wparentheses]
 1027 |     _u8 leftBytes = 4 - len & 0x3;
      |                     ~~^~~~~
src/rplidar_driver.cpp: In member function ‘virtual void rp::standalone::rplidar::RPlidarDriverImplCommon::_ultraCapsuleToNormal(const rplidar_response_ultra_capsule_measurement_nodes_t&, rplidar_response_measurement_node_hq_t*, size_t&)’:
src/rplidar_driver.cpp:1238:32: warning: comparison of integer expressions of different signedness: ‘int’ and ‘unsigned int’ [-Wsign-compare]
 1238 |             if ((dist_predict1 == 0xFFFFFE00) || (dist_predict1 == 0x1FF)) {
      |                  ~~~~~~~~~~~~~~^~~~~~~~~~~~~
src/rplidar_driver.cpp:1246:32: warning: comparison of integer expressions of different signedness: ‘int’ and ‘unsigned int’ [-Wsign-compare]
 1246 |             if ((dist_predict2 == 0xFFFFFE00) || (dist_predict2 == 0x1FF)) {
      |                  ~~~~~~~~~~~~~~^~~~~~~~~~~~~
src/rplidar_driver.cpp:1256:21: warning: unused variable ‘syncBit_check_threshold’ [-Wunused-variable]
 1256 |                 int syncBit_check_threshold = (int)((3 << 16) / angleInc_q16)+1;//find syncBit in 0~1 degree
      |                     ^~~~~~~~~~~~~~~~~~~~~~~
src/rplidar_driver.cpp: In member function ‘virtual u_result rp::standalone::rplidar::RPlidarDriverImplCommon::startMotor()’:
src/rplidar_driver.cpp:2222:26: warning: control reaches end of non-void function [-Wreturn-type]
 2222 |         setLidarSpinSpeed(600);//set default rpm to tof lidar
      |         ~~~~~~~~~~~~~~~~~^~~~~
src/rplidar_driver.cpp: In member function ‘virtual void rp::standalone::rplidar::RPlidarDriverImplCommon::_ultraCapsuleToNormal(const rplidar_response_ultra_capsule_measurement_nodes_t&, rplidar_response_measurement_node_hq_t*, size_t&)’:
src/rplidar_driver.cpp:1249:31: warning: ‘scalelvl2’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 1249 |                 dist_predict2 = (dist_predict2 << scalelvl2);
      |                 ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/rplidar_driver.cpp:1241:31: warning: ‘scalelvl1’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 1241 |                 dist_predict1 = (dist_predict1 << scalelvl1);
      |                 ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 CXX  src/hal/thread.cpp
 CXX  src/arch/linux/net_serial.cpp
src/arch/linux/net_serial.cpp: In member function ‘virtual void rp::arch::net::raw_serial::cancelOperation()’:
src/arch/linux/net_serial.cpp:428:12: warning: ignoring return value of ‘ssize_t write(int, const void*, size_t)’, declared with attribute warn_unused_result [-Wunused-result]
  428 |     ::write(_selfpipe[1], "x", 1);
      |     ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/string.h:495,
                 from /home/jacob/SourceCode/LiDAR/rplidar_sdk/sdk/sdk/src/arch/linux/arch_linux.h:40,
                 from src/arch/linux/net_serial.cpp:35:
In function ‘char* strncpy(char*, const char*, size_t)’,
    inlined from ‘virtual bool rp::arch::net::raw_serial::bind(const char*, uint32_t, uint32_t)’ at src/arch/linux/net_serial.cpp:91:12:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:106:34: warning: ‘char* __builtin_strncpy(char*, const char*, long unsigned int)’ specified bound 200 equals destination size [-Wstringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 CXX  src/arch/linux/net_socket.cpp
src/arch/linux/net_socket.cpp: In member function ‘virtual u_result rp::arch::net::StreamSocketImpl::send(const void*, size_t)’:
src/arch/linux/net_socket.cpp:438:17: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘long unsigned int’} and ‘int’ [-Wsign-compare]
  438 |         if (ans == (int)len) {
      |             ~~~~^~~~~~~~~~~
 CXX  src/arch/linux/timer.cpp
mkdir -p `dirname /home/jacob/SourceCode/LiDAR/rplidar_sdk/sdk/output/Linux/Release/librplidar_sdk.a`
 pack rplidar_driver.o->librplidar_sdk.a
 pack thread.o->librplidar_sdk.a
 pack net_serial.o->librplidar_sdk.a
[... 24 lines cut ...]
In file included from /home/jacob/SourceCode/LiDAR/rplidar_sdk/sdk/app/simple_grabber/../../sdk/include/rplidar.h:42,
                 from main.cpp:30:
/home/jacob/SourceCode/LiDAR/rplidar_sdk/sdk/app/simple_grabber/../../sdk/include/rplidar_driver.h:297:33: note: declared here
  297 |     DEPRECATED(virtual u_result ascendScanData(rplidar_response_measurement_node_t * nodebuffer, size_t count)) = 0;
      |                                 ^~~~~~~~~~~~~~
/home/jacob/SourceCode/LiDAR/rplidar_sdk/sdk/app/simple_grabber/../../sdk/include/rplidar_driver.h:44:34: note: in definition of macro ‘DEPRECATED’
   44 |         #define DEPRECATED(func) func __attribute__ ((deprecated))
      | 
```

if I write a file:

main.cpp:
```
#include <rplidar.h>
int main() {return 0;}
```
I receive:
```In file included from main.cpp:1:
rplidar.h:38:10: fatal error: hal/types.h: No such file or directory
   38 | #include "hal/types.h"
      |          ^~~~~~~~~~~~~
compilation terminated.

```

gaborszita, 2021-04-30:
same issue

VictorConscience, 2021-09-08:
@GasserJC did you manage to solve this issue by any chance ?

RickCarp, 2022-05-09:
@gaborszita , @GasserJC , @VictorConscience  did you find a solution ?

GasserJC, 2022-05-09:
@VictorConscience I believe flynn248 at this repo resolved it. 

https://github.com/JacSchn/5G-Beamforming-from-Visual-and-Lidar-Rendering

Abdulhadiasa, 2022-10-03:
I fixed it by changing 
https://github.com/Slamtec/rplidar_sdk/blob/babe753f39f16f4c6a6dec2c33b4ed541ca0e502/sdk/include/rplidar.h#L38
to 
`#include "../src/hal/types.h"
`