Slamtec/rplidar_sdk #67: Pi4 Comple Error.

neolu, 2021-12-16:
CROSS_COMPILE_PREFIX=arm-linux-gnueabihf ./cross_compile.sh
make[1]: Entering directory '/home/pi/works/rplidar_sdk/sdk'
 CXX  src/sl_lidar_driver.cpp
In file included from src/sl_lidar_driver.cpp:40:
src/hal/event.h: In member function ‘long unsigned int rp::hal::Event::wait(long unsigned int)’:
src/hal/event.h:141:31: warning: conversion from ‘rp::hal::Event::<unnamed enum>’ to ‘long unsigned int’ changes value from ‘rp::hal::Event::EVENT_TIMEOUT’ to ‘4294967295’ [-Woverflow]
  141 |                         ans = EVENT_TIMEOUT;
      |                               ^~~~~~~~~~~~~
src/sl_lidar_driver.cpp: In member function ‘virtual sl_result sl::SlamtecLidarDriver::grabScanDataHq(sl_lidar_response_measurement_node_hq_t*, size_t&, sl_u32)’:
src/sl_lidar_driver.cpp:556:34: error: narrowing conversion of ‘rp::hal::Event::EVENT_TIMEOUT’ from ‘long long unsigned int’ to ‘long unsigned int’ [-Wnarrowing]
  556 |             case rp::hal::Event::EVENT_TIMEOUT:
      |                                  ^~~~~~~~~~~~~
src/sl_lidar_driver.cpp:556:13: warning: conversion from ‘long long unsigned int’ to ‘long unsigned int’ changes value from ‘18446744073709551615’ to ‘4294967295’ [-Woverflow]
  556 |             case rp::hal::Event::EVENT_TIMEOUT:
      |             ^~~~
src/sl_lidar_driver.cpp: In member function ‘sl_result sl::SlamtecLidarDriver::_sendCommand(sl_u16, const void*, size_t)’:
src/sl_lidar_driver.cpp:1089:35: warning: comparison of integer expressions of different signedness: ‘int’ and ‘std::vector<unsigned char>::size_type’ {aka ‘unsigned int’} [-Wsign-compare]
 1089 |             for (int pos = 0; pos < cmd_packet.size(); pos++) {
      |                               ~~~~^~~~~~~~~~~~~~~~~~~
src/sl_lidar_driver.cpp:1058:19: warning: unused variable ‘pkt_header’ [-Wunused-variable]
 1058 |             sl_u8 pkt_header[10];
      |                   ^~~~~~~~~~
src/sl_lidar_driver.cpp: In member function ‘void sl::SlamtecLidarDriver::_ultraCapsuleToNormal(const sl_lidar_response_ultra_capsule_measurement_nodes_t&, sl_lidar_response_measurement_node_hq_t*, size_t&)’:
src/sl_lidar_driver.cpp:1370:40: warning: comparison of integer expressions of different signedness: ‘int’ and ‘unsigned int’ [-Wsign-compare]
 1370 |                     if ((dist_predict1 == 0xFFFFFE00) || (dist_predict1 == 0x1FF)) {
      |                          ~~~~~~~~~~~~~~^~~~~~~~~~~~~
src/sl_lidar_driver.cpp:1379:40: warning: comparison of integer expressions of different signedness: ‘int’ and ‘unsigned int’ [-Wsign-compare]
 1379 |                     if ((dist_predict2 == 0xFFFFFE00) || (dist_predict2 == 0x1FF)) {
      |                          ~~~~~~~~~~~~~~^~~~~~~~~~~~~
src/sl_lidar_driver.cpp: At global scope:
src/sl_lidar_driver.cpp:72:17: warning: ‘void sl::convert(const sl_lidar_response_measurement_node_hq_t&, sl_lidar_response_measurement_node_t&)’ defined but not used [-Wunused-function]
   72 |     static void convert(const sl_lidar_response_measurement_node_hq_t& from, sl_lidar_response_measurement_node_t& to)
      |                 ^~~~~~~
src/sl_lidar_driver.cpp:59:17: warning: ‘void sl::printDeprecationWarn(const char*, const char*)’ defined but not used [-Wunused-function]
   59 |     static void printDeprecationWarn(const char* fn, const char* replacement)
      |                 ^~~~~~~~~~~~~~~~~~~~
make[1]: *** [/home/pi/works/rplidar_sdk/mak_common.inc:82: /home/pi/works/rplidar_sdk/obj/Linux/Release/sdk/src/sl_lidar_driver.o] Error 1
make[1]: Leaving directory '/home/pi/works/rplidar_sdk/sdk'
make: *** [/home/pi/works/rplidar_sdk/mak_common.inc:43: make_subs] Error 1
pi@raspberrypi:~/works/rplidar_sdk $

bramburn, 2022-01-07:
do you have the gnu compiler installed and pointed correctly? are you building on a linux box or WSL?

ARDev1161, 2022-01-10:
> do you have the gnu compiler installed and pointed correctly? are you building on a linux box or WSL?
This can help https://github.com/Slamtec/rplidar_sdk/pull/66/files