Slamtec/rplidar_sdk #71: Compiler error LNK2019

Programmer54, 2022-06-08:
After making some very minor changes (which I had to do to get the sdk to work), when I try to compile I get this error:

1:
Severity Code Description Project File Line Hide Status
Error LNK2019 unconverted outer symbol "struct sl::Result<class sl::IChannel *> __cdecl sl::createSerialPortChannel(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,int)" (?createSerialPortChannel@sl@@YA?AU?$Result@PEAVIChannel@sl@@@1@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D @2@@std@@H@Z) referenced at your parent ConsoleApplication1

2: 
Severity Code Description Project File Line Hide Status
Error LNK2019 unresolved external symbol "struct sl::Result<class sl::ILidarDriver *> __cdecl sl::createLidarDriver(void)" (?createLidarDriver@sl@@YA?AU?$Result@PEAVILidarDriver@sl@@@1@ XZ) referenced in main function ConsoleApplication1 

NOTE:
The examples that come with the sdk are not compiled, but the project required for the .lib file is compiled. So, I have a compiled file "rplidar_driver.lib" and the LIBRARY setting has been made for the project.

Ajasra, 2023-01-12:
Do you solve this problem? I have the same. Looks like the version of the rplidar_driver and linking it.

Programmer54, 2023-01-12:
I stopped looking at this project a long time ago. But I think I solved this problem by setting the generated code attribute when linking the library. as follows:
Project properties > Refinement > Generate Code During Linking => I set this property but I couldn't remember which one I chose. You can learn this value from the sample project in the lidar sdk. I hope I was helpful, if your problem persists please feel free to write :) Good work

Ajasra, 2023-01-13:
Solved. There are a few steps. The Visual Studio configuration manager can get into some trouble with it.
1. Add the '**rplidar_driver**' from the SDK into your solution as dependence.
   - _Solution - Add - Existing project_
   - _Solution - Dependency - select '**rplidar_driver**" project.
   [image]
2. _Properties - C/C++ - General - Additional include directories_ (add the directory with /sdk,  /sdk/include, /sdk/src in relation to your folder structure) 
[image]
4. _Properties - Linker - General - Additional Library Directories_ (add here the path to where 'rplidar_driver' build .lib file)
[image]
6. _Properties - Linker - Input - Additional Dependencies_ (add 'rplidar_driver.lib')
[image]
8. _Properties - C/C++ - Code Generator - Runtime library_ (be sure both your project and rplidar_driver have the same option there, I used /MTd)
[image]
10. Now we need to change the rplidar_driver to x64; in the configuration manager, there are bugs; it does not let you create an x64 solution for rplidar_driver if you already have x64 dilution in your project (say a solution with this name already exist). So, I need to delete the x63 option first and after adding it again everywhere)
   - _Properties - Configuration manager_
   - delete x64 platform for all projects and delete the x64 solution
   [image]
   - add x64 platform to rplidar_driver, add x64 platform to your project (uncheck 'create a new solution platform)
   [image]
   - add x64 solution to '**Active solution platform**'
   [image]
   - select all projects to have the same platform (for me, I needed to do this process two times. The first time x64 just did not appear in the list after I added it. So I needed to delete all x64 platforms and solutions and restart Visual Studio)
  [image]

Then I was able to build both driver and my project in x64 without errors.
Thanks to [Tamaya31](https://github.com/Tamaya31) for help and direction to look into.