Slamtec/rplidar_sdk #72: Cannot run ultra_simple

Lagree007, 2022-06-08:
[image]

I keep getting the attached when I try to run the ultra_simple. I followed the instruction on the Github page.

TonioBoubs, 2022-06-13:
I have the same problem with the A1M8 lidar ...
The same information are written in the terminal but no way to read and print some measurement ...
I tried to comment the reading of args in the main.cpp file and hard write this 👍 
```
opt_channel_param_first = "/dev/ttyUSB0";
opt_channel_param_second = 115200;
```

Juggernaut101DK, 2022-06-14:
@Lagree007 
I think the problem is, that the SDK documentation has not been updated.

To use ultra_simple or simple_grabber the command is different now.

Try: 
```console
./ultra_simple --channel --serial /dev/ttyUSB0 115200
````
or  
```console
sudo ./ultra_simple --channel --serial /dev/ttyUSB0 115200
````

You should also check that the baudrate on the USB2UART adapter, which connects the Lidar with the computer, is set accordingly to either:

- A2 => 115200
- A3 => 256000

You can also check, if you are using the correct COM PORT on Linux by quickly checking it with:
```console
dmesg | grep ttyUSB
````
[image]

Lagree007, 2022-06-14:
@Juggernaut101DK  Thanks it worked.

livethisdream, 2022-11-07:
> @Lagree007 I think the problem is, that the SDK documentation has not been updated.
> 
> To use ultra_simple or simple_grabber the command is different now.
> 
> Try:
> 
> ```
> ./ultra_simple --channel --serial /dev/ttyUSB0 115200
> ```
> 
> or
> 
> ```
> sudo ./ultra_simple --channel --serial /dev/ttyUSB0 115200
> ```
> 
> You should also check that the baudrate on the USB2UART adapter, which connects the Lidar with the computer, is set accordingly to either:
> 
>     * A2 => 115200
> 
>     * A3 => 256000
> 
> 
> You can also check, if you are using the correct COM PORT on Linux by quickly checking it with:
> 
> ```
> dmesg | grep ttyUSB
> ```
> 
> [image]

For others to reference, the baudrate for the S2 is 1000000.

rishichoco, 2024-08-07:
> [image]
> 
> I keep getting the error:
> 
> "Ultra simple LIDAR data grabber for SLAMTEC LIDAR. Version: SL_LIDAR_SDK_VERSION Simple LIDAR data grabber for SLAMTEC LIDAR. Version: SL_LIDAR_SDK_VERSION Usage: For serial channel ./ultra_simple --channel --serial [baudrate] The baudrate is 115200(for A2) or 256000(for A3). For udp channel ./ultra_simple --channel --udp [port NO.] The LPX default ipaddr is 192.168.11.2,and the port NO.is 8089. Please refer to the datasheet for details."
> 
> when I try to run the ultra_simple. I followed the instruction on the Github page. I am using the A2M8 lidar with the USB adapter "CP2102 USB to UART Bridge Controller"

you need to give read and write access RPlidar for that you need run this this command `sudo chmod 666 /dev/ttyUSB0` then you can run this command` ./ultra_simple --channel --serial /dev/ttyUSB0 115200`