Slamtec/rplidar_sdk #78: Compile errors on Ubuntu 22.04

c1505, 2022-07-13:
## System Information
* versions tried: v1.12.0 and v1.11.0 (logs are from 1.12)
* Ubuntu 22.04
* g++ version 11.2.0

## To reproduce
* Download release from here https://github.com/Slamtec/rplidar_sdk/archive/refs/tags/release/v1.12.0.tar.gz or 1.11 from here https://www.slamtec.com/en/Support#rplidar-a-series 
* extract
* navigate to sdk directory
* run `make` or `sudo make`

## Logs 
```bash
make[1]: Entering directory '/home/corey/rplidar_sdk-release-v1.12.0/sdk/sdk'
 CXX  src/rplidar_driver.cpp
src/rplidar_driver.cpp: In function ‘_u32 rp::standalone::rplidar::_crc32cal(_u32, void*, _u16)’:
src/rplidar_driver.cpp:997:23: warning: suggest parentheses around ‘-’ in operand of ‘&’ [-Wparentheses]
  997 |     _u8 leftBytes = 4 - len & 0x3;
      |                     ~~^~~~~
src/rplidar_driver.cpp: In member function ‘virtual void rp::standalone::rplidar::RPlidarDriverImplCommon::_ultraCapsuleToNormal(const rplidar_response_ultra_capsule_measurement_nodes_t&, rplidar_response_measurement_node_hq_t*, size_t&)’:
src/rplidar_driver.cpp:1208:32: warning: comparison of integer expressions of different signedness: ‘int’ and ‘unsigned int’ [-Wsign-compare]
 1208 |             if ((dist_predict1 == 0xFFFFFE00) || (dist_predict1 == 0x1FF)) {
      |                  ~~~~~~~~~~~~~~^~~~~~~~~~~~~
src/rplidar_driver.cpp:1216:32: warning: comparison of integer expressions of different signedness: ‘int’ and ‘unsigned int’ [-Wsign-compare]
 1216 |             if ((dist_predict2 == 0xFFFFFE00) || (dist_predict2 == 0x1FF)) {
      |                  ~~~~~~~~~~~~~~^~~~~~~~~~~~~
src/rplidar_driver.cpp: In member function ‘virtual u_result rp::standalone::rplidar::RPlidarDriverImplCommon::grabScanData(rplidar_response_measurement_node_t*, size_t&, _u32)’:
src/rplidar_driver.cpp:1795:26: error: narrowing conversion of ‘rp::hal::Event::EVENT_TIMEOUT’ from ‘int’ to ‘long unsigned int’ [-Wnarrowing]
 1795 |     case rp::hal::Event::EVENT_TIMEOUT:
      |                          ^~~~~~~~~~~~~
src/rplidar_driver.cpp: In member function ‘virtual u_result rp::standalone::rplidar::RPlidarDriverImplCommon::grabScanDataHq(rplidar_response_measurement_node_hq_t*, size_t&, _u32)’:
src/rplidar_driver.cpp:1824:26: error: narrowing conversion of ‘rp::hal::Event::EVENT_TIMEOUT’ from ‘int’ to ‘long unsigned int’ [-Wnarrowing]
 1824 |     case rp::hal::Event::EVENT_TIMEOUT:
      |                          ^~~~~~~~~~~~~
src/rplidar_driver.cpp: In member function ‘virtual u_result rp::standalone::rplidar::RPlidarDriverImplCommon::startMotor()’:
src/rplidar_driver.cpp:2190:26: warning: control reaches end of non-void function [-Wreturn-type]
 2190 |         setLidarSpinSpeed(600);//set default rpm to tof lidar
      |         ~~~~~~~~~~~~~~~~~^~~~~
make[1]: *** [/home/corey/rplidar_sdk-release-v1.12.0/sdk/mak_common.inc:82: /home/corey/rplidar_sdk-release-v1.12.0/sdk/obj/Linux/Release/sdk/src/rplidar_driver.o] Error 1
make[1]: Leaving directory '/home/corey/rplidar_sdk-release-v1.12.0/sdk/sdk'
make: *** [/home/corey/rplidar_sdk-release-v1.12.0/sdk/mak_common.inc:43: make_subs] Error 1
```

youngpad, 2022-09-04:
I am experiencing the same issue.

FlabioMirelez, 2022-10-21:
Similar error:
CXX  src/arch/linux/net_socket.cpp
src/arch/linux/net_socket.cpp: In member function ‘virtual u_result rp::net::SocketAddress::getAddressAsString(char*, size_t) const’:
src/arch/linux/net_socket.cpp:170:15: error: ordered comparison of pointer with integer zero (‘const char*’ and ‘int’)
  170 |     return ans<=0?RESULT_OPERATION_FAIL:RESULT_OK;

livethisdream, 2022-11-07:
@roleohibachi and I worked this out and we're not sure if it's the right solution, but it's *a* solution that allows the code to compile.
In sdk/src/arch/linux/net_socket.cpp on line 151 and 171, change:
    return ans<=0?RESULT_OPERATION_FAIL:RESULT_OK;
to:
    return ans!=0?RESULT_OPERATION_FAIL:RESULT_OK;

and recompile from the root directory.

ajuton-ens, 2023-02-23:
2 problems of data types :
src/sl_lidar_driver.cpp:556:34: error: narrowing conversion of ‘rp::hal::Event::EVENT_TIMEOUT’ from ‘int’ to ‘long unsigned int’ [-Wnarrowing]
is solved by adding an explicit cast in the file sl_lidar_driver.cpp line 556 :
```
case (long unsigned int)rp::hal::Event::EVENT_TIMEOUT:
case (long unsigned int)rp::hal::Event::EVENT_OK:
```

src/arch/linux/net_socket.cpp:170:15: error: ordered comparison of pointer with integer zero (‘const char*’ and ‘int’)
is solved by adding an explicit cast in the file net_socket.cpp line 170 :
```
return (ans<=(const char *)0)?RESULT_OPERATION_FAIL:RESULT_OK;
```