Slamtec/rplidar_sdk #84: Running server for RoboStudio to connect to?

javadan, 2022-08-30:
Hi everyone

After a bit of fuss, I've got the SDK compiling and building the sample apps on my RPi Zero W, with an RP-A1M8-R6, and getting some output from 
`~/rplidar_sdk/output/Linux/Release $ ./simple_grabber --channel --serial /dev/ttyUSB0 115200`

Wondering if anyone's ever managed to connect the Slamtec RoboStudio to a TCP or UDP port of another computer in the LAN?  
Is that just for "SLAMWare" devices? Is there any useful documentation? (The user manual is not useful).

I see there's some files sl_tcp_channel.cpp, sl_udp_channel.cpp.  
It's nice that there's some server code, but it would be pretty cool if there was an basic example, too, like, using the protocol, etc.

I would imagine that connecting the RoboStudio to a headless device running an RPLidar would be a common use case.  (?)

From what I can see, I'm probably just going to end up hacking simple_grabber, publish my own data stream of "theta / dist" values, and write my own visualiser on the other side, by the looks of it.

datino115, 2022-11-01:
May I ask what you did to get the SDK to compile on your rp-A1M8 lidar? I'm still stuck trying to fix the files.

javadan, 2022-11-01:
Hi,  I got it compiling after changing this bit https://github.com/Slamtec/rplidar_sdk/pull/66/files