Slamtec/rplidar_sdk #88: *** buffer overflow detected ***: ./ultra_simple terminated

tcexeexe, 2022-10-29:
Hello~
 I use order `./ultra_simple --channel -s ttyUSB0 256000` to test the RPLIDAR S1, but en error occurred that 
```
Ultra simple LIDAR data grabber for SLAMTEC LIDAR.
Version: SL_LIDAR_SDK_VERSION
*** buffer overflow detected ***: ./ultra_simple terminated
Aborted (core dumped)
```
l have had use the order 'sudo chmod a+rw /dev/ttyUSB0' to set the UART permission

livethisdream, 2022-11-07:
I am using the S2 and had to run it as sudo and also specify the baudrate as 1000000 -- this is specific to the LiDAR you're using.

LDec-AlmostHuman, 2023-02-15:
Hi tcexeexe,

Did you manage to resolve the problem?
I am facing the same problem with A2.
Using ./ultra_simple --channel --serial ttyUSB0 115200 in my case.
Throwing the same error as you have.

Anyone with solution?

LDec-AlmostHuman, 2023-03-08:
If you are still having problems please let me know and I will find my solution for it. Took me two days but problem solved.

LDec-AlmostHuman, 2023-03-08:
But cannot help with S2 model as I have only A2M12, so via USB.

totenkatze, 2023-03-10:
@LDec-AlmostHuman I'm jumping into the thread because I have an A2M12 as well and have been getting the buffer overflow error. Although, I seemed to have reached a state where I am back to a "cannot bind to the specified serial port /dev/ttyUSB0" error that I had been experiencing before.

If I'm still having issues after I get through that, can I call on you to dig up the solution you found?

LDec-AlmostHuman, 2023-03-10:
> @LDec-AlmostHuman I'm jumping into the thread because I have an A2M12 as well and have been getting the buffer overflow error. Although, I seemed to have reached a state where I am back to a "cannot [...]

I will do that on Sunday, the latest if you don't mind. My setup was Ubuntu 20.04 and not in docker. Having problems with docker exposing /dev even with --privileged, but when I will fix that issue I will post as well docker version of it.

totenkatze, 2023-03-11:
I was able to get this working on my raspberry pi 4 running ubuntu 22.04 without having to chmod /dev/ttyUSB0 or add myself to the dialout group. I did have to run  "./ultra_simple --channel --serial /dev/ttyUSB0 256000" as sudo though as a result.

I'm still having issues getting it to work on my jetson nano, which is running xubuntu 20.04. I'll have to look into where I went wrong there.

LDec-AlmostHuman, 2023-03-11:
@totenkatze
Please see below fix for Ubuntu 20.04 & Ubuntu 22.04.
I have tested it on Jetson NX with Ubuntu 20.04 and AMD PC with both Ubuntu 20.04 & Ubuntu 22.04.

# Tested on Ubuntu 20.04 without problems.
====================================================
# On Ubuntu 22.04, after getting 
git clone https://github.com/Slamtec/rplidar_sdk.git
# you will need to correct line 175 in net_socket.cpp, which is in rplidar_sdk/sdk/src/arch/linux
# Replace:
return ans<=0?RESULT_OPERATION_FAIL:RESULT_OK;
# with:
return ans == nullptr ? RESULT_OPERATION_FAIL : RESULT_OK;
# save the file, get back to rplidar_sdk directory and now should compile.
====================================================
# I have not tested it to see if it works on MacOS (Silicone)
# I have not tested it to see if it works on any Windows flavors.
====================================================

# RPLidar A2M12, which has 256000 communication speed.
# Change the speed to 256000 with the switch on the device.

# Check if you can see CP2102 USB to UART Bridge Controller
# Run:
usb-devices

# It should return something like this, that means that the driver is loaded.
# If not, look a few lines below for installation instructions from Slamtec and start from the beginning.

T:  Bus=05 Lev=01 Prnt=01 Port=02 Cnt=01 Dev#=  4 Spd=12  MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=10c4 ProdID=ea60 Rev=01.00
S:  Manufacturer=Silicon Labs
S:  Product=CP2102 USB to UART Bridge Controller
S:  SerialNumber=0001
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=cp210x
E:  Ad=01(O) Atr=02(Bulk) MxPS=  64 Ivl=0ms
E:  Ad=81(I) Atr=02(Bulk) MxPS=  64 Ivl=0ms

# Run below line to see ownership and which ttyUSBx you see
ls -l /dev | grep ttyUSB

# It should return something like this:
crw-rw----   1 root  dialout 188,     0 Mar 11 06:44 ttyUSB0

# Change permissions to that device (others are suggesting using 666, but for me only 777 worked):
sudo chmod 666 /dev/ttyUSB0
sudo chmod 777 /dev/ttyUSB0

# Run again:
ls -l /dev | grep ttyUSB

# Now should return changed permission:
crwxrwxrwx   1 root  dialout 188,     0 Mar 11 06:44 ttyUSB0

# Add current user to dialout:
sudo adduser $USER dialout

# Installation from Slamtec to install drivers
# If usd-devices doesn't return any info on Product=CP2102 USB to UART Bridge Controller - to install drivers.
# You will need to start the above procedure from the beginning.
git clone https://github.com/Slamtec/rplidar_sdk.git
cd rplidar_sdk

# Install ament_cmake if you don't have one installed already - my installation was missing ament-cmake and was throwing errors
sudo apt-get update
sudo apt-get -y install ament-cmake 

make
[... 60 lines cut ...]
# For more data you can press y and will show some more data and will stop.
# To stop press n and will stop.

# Now let's test ultra_simple
./ultra_simple --channel --serial /dev/ttyUSB0 257175
# It should return something like this:
========================================
theta: 0.00 Dist: 00000.00 Q: 0 
theta: 0.23 Dist: 00000.00 Q: 0 
theta: 0.47 Dist: 00000.00 Q: 0 
theta: 0.70 Dist: 00000.00 Q: 0 
theta: 0.93 Dist: 00000.00 Q: 0 
theta: 1.17 Dist: 00000.00 Q: 0 
theta: 1.40 Dist: 00000.00 Q: 0 
....
theta: 358.59 Dist: 00000.00 Q: 0 
theta: 358.83 Dist: 00000.00 Q: 0 
theta: 359.06 Dist: 00000.00 Q: 0 
theta: 359.30 Dist: 00000.00 Q: 0 
theta: 359.53 Dist: 00000.00 Q: 0 
theta: 359.76 Dist: 00000.00 Q: 0 
========================================

# I hope all works:)
# If you are interested I can post as well corrected python files for git clone https://github.com/Slamtec/sllidar_ros2.git

LDec-AlmostHuman, 2023-03-11:
@totenkatze 
I was testing it with Jetson Nano with JetPack 4.6.1 and I have managed to spin it, but couldn't see results as Nano froze and since I have a problem with powering via jack, only powering via USB seems to work. PSU is 5.2V 4A under full load and 5.22V without any load. No idea what happened, but will have time to work on it by the end of the month so will post as well Nano instructions. 
BTW are you thinking of switching to Orin Nano or Orin NX 😄? Cannot wait to get my hands on it. I was thinking of AGX Orin dev kit, but it's a bit too pricey for my current budget. Despite Orin NX with carrier board will be around 1.000€ for 16GB model. I'm still in the dark with PCIE 4 on carrier board. All available now seem to be PCIE 3. All my NVME drives are PCIE 4. I know they will work, but ... 😄.

totenkatze, 2023-03-11:
@LDec-AlmostHuman Thank you for sharing the instructions above. I'll give them a try on my nano this weekend. I have tried most of the items in the instructions without success, but I'm curious to see if adjusting the baud rate would help me out.

Getting things running so effortlessly on my Pi with 22.04 while still having such difficulty on the nano makes be feel a bit less crazy.

I haven't looked into the Orin yet. I'm just getting started with robotics as a hobby. I have a few old Raspberry Pis, two Jetson Nanos, and an Orange Pi 5 on the way. With the $XXXX dollars I've spent on robots and parts, I think that I should have just starting doing cocaine to save a bit of money.

LDec-AlmostHuman, 2023-03-11:
@totenkatze 
Maybe Nano has not enough juice to spin the Lidar. 
Try to connect power to Lidar to other source of power. I can try to install Docker image of Xubuntu 20.04 on Jetson NX and try to see if it works. I don't have second Nano. And will have time to fix Nano in circa 2 weeks. 
You are not the only one with $xxxx spent on robotics hobby projects 😄. And I actually like doing such projects. 
Next week I was thinking of running both A2M12 and OAK-D S2 to see how would that setup work.

LDec-AlmostHuman, 2023-03-12:
@totenkatze 
Just tested on Jetson Nano with Xubuntu 20.04 and works without problems. I have followed instructions on installing Xubuntu from forums.developer.com
Can send you link to it. All together took me a bit over an hour, but all works. Lidar is spinning etc.

LDec-AlmostHuman, 2023-03-13:
Anyone succeed with my guidance? If not please let me know.

LDec-AlmostHuman, 2023-03-14:
@totenkatze 
They are confusing B01 😄. Boots from SDCard without issues. It hangs on USB, but I think I know what's the problem is. I have all other devices plugged in and it seems I need to boot for the first time with only USB drive plugged in. Didn't do it yet, just found out that from one of the posts. Let me know if you succeeded with RPLidar.

LDec-AlmostHuman, 2023-03-14:
@totenkatze 
Someone wrote on NVidia forum that there might be some issues on B01 when you are using A01 in extlinux. Maybe this is the problem.

totenkatze, 2023-03-14:
@LDec-AlmostHuman I started from fresh install of xubuntu and installed git, make, g++, etc. and after the compile, it worked when running as sudo. I chmod /dev/ttyUSB0 to 666 (any form of XX6 works) and it worked without sudo privileges. Now I have everything working on both my Pi and Nano =).