Slamtec/rplidar_sdk #101: how to increase the rplidar S2 publishing frequency upto 15Hz?

kbln, 2023-04-30:
how to increase the rplidar S2 publishing frequency upto 15Hz?
Rplidar current publish frequency is only 10Hz. However the publishing frequency of rplidar A2 is 13 to 15 Hz.

WubinXia, 2023-05-05:
The typical frequency of RPLIDAR S2 is 10Hz, if you want to increase frequency,  please make sure the input power >5V, and follow this steps,
1. Connecting to Rplidar 
2.  Starting scan
3. After grab first scan ,  calling  the function  startMotor(15)

kbln, 2023-05-05:
Thanks for the answer! 
But I got an error when trying to compile the ultra_simple (class "sl::ILidarDriver" has no member "startMotor")
How to fix it?
[image]