Slamtec/rplidar_sdk #112: Error, cannot bind to the specified serial port /dev/ttyUSB0 on ubuntu 20.04

KianRK, 2023-07-24:
I know there are already issues concering this issue, but none of the suggested solutions worked for me.

I am trying to run the ultra_simple demo with the following command:
...rplidar_sdk-master/output/Linux/Release$ sudo ./ultra_simple --channel --serial /dev/ttyUSB0 115200

I tried it with different baud rates. The device is recognized:
"dmesg | ttyUSB0" : usb 1-1: cp210x converter now attached to ttyUSB0

I added $USER to dialout even though I run it with sudo anyways, but when I run without sudo I get the same error.

I also changed permissions to 777. Nothing works.

I would be very thankful for help.

deyouslamtec, 2023-08-09:
the baudrate used by different models is as follows:
  A1(115200),A2M7(256000),A2M8(115200),A2M12(256000),A3(256000),S1(256000),S2(1000000),S3(1000000)

KianRK, 2023-08-10:
It does run on 256000 even though the manual says otherwise. But that was not the Problem in the first place, since I tried it several times with different baud rates. I hat to plug the power input to a phone charger to give it sufficient power and then it worked. After that it also worked when I plugged it back to my laptop. I don't know why it is this way, but that solved my problem.

closed 2023-08-17 by deyouslamtec