Slamtec/rplidar_sdk #114: S2 motor speed control

aalex, 2023-07-26:
S2 motor speed control

How can change the motor to something like 15 Hz instead of the default 10 Hz?

Here is the output of `getMotorInfo`:
- max_speed: 1200
- min_speed: 480
- desired_speed: 600

But no matter what value I provide by calling `setMotorSpeed`, the `desired_speed` remains at `600`.
(600 seems to mean 600 RPM, which is 10 Hz)

```cpp
/// Attempt to set the motor speed:
if (opt_channel_type == CHANNEL_TYPE_SERIALPORT) {
            std::cout << "Set motor speed to " << settings.get_motor_speed() << std::endl;
            //drv->setMotorSpeed();
            // Trying to change the motor speed:
            sl_result setSpeedResult = drv->setMotorSpeed(settings.get_motor_speed());
            std::cout << "Set speed? ";
            switch (setSpeedResult) {
            case SL_RESULT_OK:
                std::cout << "ok" << std::endl;
                break;
            case SL_RESULT_OPERATION_FAIL:
                std::cout << "fail" << std::endl;
                break;
            case SL_RESULT_OPERATION_NOT_SUPPORT:
                std::cout << "not support" << std::endl;
                break;
            case SL_RESULT_OPERATION_STOP:
                std::cout << "stop" << std::endl;
                break;
            case SL_RESULT_OPERATION_TIMEOUT:
                std::cout << "timeout" << std::endl;
                break;
            }
            LidarMotorInfo motorInfo;
            drv->getMotorInfo(motorInfo);
            std::cout << "max_speed: " << motorInfo.max_speed << std::endl;
            std::cout << "min_speed: " << motorInfo.min_speed << std::endl;
            std::cout << "desired_speed: " << motorInfo.desired_speed << std::endl;
            std::cout << "motorCtrlSupport: " << motorInfo.motorCtrlSupport << std::endl; //0=none, 1=pwn, 2=rpm
        }
```

kde12327, 2024-05-08:
Is this problem solved?
I got same problem on A1 model.
Result of setMotorSpeed() is SL_RESULT_OK.
But desired_speed doesn't changed.

delessertyanis-ux, 2026-08-21:
I had the same issue in python, i had to change the default speed value of the motor directly in the library. Maybe it works in C/C++ in the sl_lidar_cmd.h or sl_lidar_driver.h library.