Slamtec/rplidar_sdk #116: unable to make the SDK

r8drfn89, 2023-08-16:
[image]

I cloned the repository for the sdk. Then in the root of the sdk I tried to run make. These are the errors I get. 

sudo make and make DEBUG=1 also fail and give this same errors

r8drfn89, 2023-08-16:
I even tried it in the rplidar_sdk/sdk       still gives the same error
[image]

deyouslamtec, 2023-08-17:
Try dev-sdk branch, the latest commit may solve this problem

r8drfn89, 2023-08-17:
@deyouslamtec What do you mean dev-sdk branch?

deyouslamtec, 2023-08-17:
[image]

r8drfn89, 2023-08-17:
Okay thanks for showing where to click. Still does not explain what to do. Am I supposed to clone from that branch?

deyouslamtec, 2023-08-17:
yes. clone or download dev-sdk branch

r8drfn89, 2023-08-17:
[image]
@deyouslamtec I tried what you suggested and the same error is occurring during the make.

deyouslamtec, 2023-08-18:
please run `git clone -b dev-sdk https://github.com/Slamtec/rplidar_sdk.git` to clone dev-sdk branch

r8drfn89, 2023-08-19:
I have successfully downloaded the sdk and went through the make. Now I am getting an error trying to run the applications. 
[image]
Where am I supposed to run the demo application?

deyouslamtec, 2023-08-21:
The APP executable file is in the rplidar_sdk/output/Linux/Release/ directory

closed 2023-08-23 by r8drfn89