Slamtec/rplidar_sdk #120: sdk no longer working

egonzalez1958, 2023-09-16:
After failing to work with sdk using ubuntu 22.04.3 arm64 on a raspberry pi 4B using an A2-M12, I tried replacing the raspberry for a laptop using an i7, 7th generation with ubuntu 22.04.3 amd64.
The codes of custom_baudrate, simple_grabber and ultra_simple were compiled without errors, but when trying to execute them, I got this kind of failures:

~/claser/rplidar_sdk-master/output/Linux/Release$  ./custom_baudrate /dev/ttyUSB0 256000
Baudrate negotiation demo for SLAMTEC LIDAR.
Try to establish communication to the LIDAR using the baudrate at /dev/ttyUSB0@256000 ...
*** buffer overflow detected ***: terminated
Aborted (core dumped)

The installation makes use of the following:

~/claser/rplidar_sdk-master/output/Linux/Release$ cat /etc/os-release
PRETTY_NAME="Ubuntu 22.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.3 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

~/claser/rplidar_sdk-master/output/Linux/Release$ g++ --version
g++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

~/claser/rplidar_sdk-master/output/Linux/Release$ ls -al /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 0 sep 16 16:42 /dev/ttyUSB0

Any idea on what's going on?

deyouslamtec, 2023-09-19:
~/claser/rplidar_sdk-master/output/Linux/Release$ ls -al /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 0 sep 16 16:42 /dev/ttyUSB0

Please add read and write permissions to /dev/ttyUSB0
example：
sudo chmod 666 /dev/ttyUSB0

egonzalez1958, 2023-09-19:
Thanks, now custom_baudrate works!

But as you see in the ensuing listing, ultra_simple seems to work but... it does not make the lidar to rotate.

> sudo chmod 666 /dev/ttyUSB0

> ./custom_baudrate /dev/ttyUSB0 257175
Baudrate negotiation demo for SLAMTEC LIDAR.
Try to establish communication to the LIDAR using the baudrate at /dev/ttyUSB0@257175 ...
The baudrate detected by the pair is 257175 bps. Error: 0.000 %
Wow, we just communicate with the LIDAR using non-standard baudrate : 257175!.
SLAMTEC LIDAR S/N: B1C999F6C9E59AD4C5E59CF74D1B3414
Firmware Ver: 1.32
Hardware Rev: 6

> ./ultra_simple --channel --serial /dev/ttyUSB0 256000
Ultra simple LIDAR data grabber for SLAMTEC LIDAR.
Version: SL_LIDAR_SDK_VERSION
SLAMTEC LIDAR S/N: B1C999F6C9E59AD4C5E59CF74D1B3414
Firmware Ver: 1.32
Hardware Rev: 6
SLAMTEC Lidar health status : 0

> ./ultra_simple --channel --serial /dev/ttyUSB0 257175
Ultra simple LIDAR data grabber for SLAMTEC LIDAR.
Version: SL_LIDAR_SDK_VERSION
SLAMTEC LIDAR S/N: B1C999F6C9E59AD4C5E59CF74D1B3414
Firmware Ver: 1.32
Hardware Rev: 6
SLAMTEC Lidar health status : 0

I haven't make any changes to the code.

Regards!

egonzalez1958, 2023-09-28:
Finally, the problem was singled out by comparing behaviour of two lidars, one of them didn't rotate with the same compiled code. After breaking the guarantee seals and opening the enclosure, a very tiny wire was found cut. Careful soldering solved the problem. Lidar is rotating. That also means that the health status of the Lidar does only a shallow check.