Slamtec/rplidar_sdk #127: C1 does not work with ultra_simple tests

alfem, 2024-03-13:
Hi!

I am testing a brand new C1 Lidar.

I downloaded master branch today, compiled it in a Jetson Orin (Ubuntu 20.04) and then 

`sudo ./ultra_simple --channel --serial /dev/ttyUSB0`

After a second, this message comes up:
```

Ultra simple LIDAR data grabber for SLAMTEC LIDAR.
Version: 2.1.0
Error, cannot bind to the specified serial port /dev/ttyUSB0.

```
Lidar does nothing at all (no noise, buzzs...). Only a green light in the serial-USB board lights up.

Any clue?

alfem, 2024-03-13:
Tested in windows, with robotstudio, and lidar works. Also tested sdk in a linuxmint laptop, and does not work.

ss32, 2024-03-29:
It's trying to bind to an IP and it doesn't appear that they've added support for serial ports.  `custom_baudrate` is the only app that works, and only if you pass it the default baud of 460800

https://github.com/Slamtec/rplidar_sdk/issues/128

ss32, 2024-03-29:
I was able to get the sample apps working by applying the patch in https://github.com/Slamtec/rplidar_sdk/pull/125 and running `custom_baudrate` *first*

```
$ ./output/Linux/Debug/custom_baudrate /dev/ttyUSB0 460800
Baudrate negotiation demo for SLAMTEC LIDAR.
Try to establish communication to the LIDAR using the baudrate at /dev/ttyUSB0@460800 ...
The baudrate detected by the pair is 0 bps. Error: 100.000 %
Warning, actual BPS error is too large. Communication based on this baudrate may fail.
Wow, we just communicate with the LIDAR using non-standard baudrate : 460800!.
SLAMTEC LIDAR S/N: 706AE1F4C2E398C4BCEA9AF3FEB14802
Model ID: 65

Firmware Ver: 1.01
Hardware Rev: 18
```

```
$ ./output/Linux/Debug/simple_grabber --channel -s /dev/ttyUSB0 460800
Try to connect to the /dev/ttyUSB0 with the baudrate at 460800...
SLAMTEC LIDAR S/N: 706AE1F4C2E398C4BCEA9AF3FEB14802
Version:  SL_LIDAR_SDK_VERSION 
Firmware Ver: 1.01
Hardware Rev: 18
Lidar health status : OK. (errorcode: 0)
waiting for data...
                                                                           
                                                                           
                                                                           
                                                                           
                                                                           
                                                                           
                                                                           
                                                                           
                                                                           
                                                                           
                                                                           
                                                                           
                                                                           
                                                                           
                                                                           
                                                                           
                                                                           
                                                                         **
        **   **  **************************************                  **
***************************************************************************
---------------------------------------------------------------------------
```

jayant07-yb, 2025-08-22:
>It's trying to bind to an IP and it doesn't appear that they've added support for serial ports. custom_baudrate is the only app that works, and only if you pass it the default baud of 460800

https://github.com/Slamtec/rplidar_sdk/issues/128

this helped me, 
braud rate for C1M2 is 460800
I think we need to update the docs.